markqvist/LibAPRS #12: support for mega2560

daskalovst, 2018-06-04:
Is somebody knows how to remap the modem pins to be possible to use it into the port H on mega2566

markqvist, 2018-06-04:
I currently don't have an atmega2560 based board myself to test it on, but you should be able to get it running by remapping the port and also modifying the ADC setup part to suit the 2560.

I might implement this at some point, but honestly it's not something that is high on the list of priorities right now.

If you figure it out, I'd be very happy to merge a pull request with it added!

daskalovst, 2018-06-04:
if you give me some advice how to remap the port i can try to make it work on the 2566.

markqvist, 2018-06-07:
Take a look in device.h, it's all there:

`// Port settings
#if TARGET_CPU == m328p
    #define DAC_PORT PORTD
    #define DAC_DDR  DDRD
    #define LED_PORT PORTB
    #define LED_DDR  DDRB
    #define ADC_PORT PORTC
    #define ADC_DDR  DDRC
#endif`

You'll also need to tune the ADC initialisation and timing.

closed 2018-06-07 by markqvist

daskalovst, 2018-06-09:
Hello, i found library for mega2560 https://github.com/williamhemmingsen/LibAPRS with this code modem pins don't need to be in order.... perfect for my case.
I can confirm this code works, but some hardware changes have to be done on the modem schematics, with original scheme output was ~0V65, my radio needs 3V3, also sinus form is not perfect but every packet is received from i-Gate...

#if TARGET_CPU == m2560
    #define DAC_PORT PORTH
    #define DAC_DDR  DDRH
    #define DAC_PINA _BV(2)
    #define DAC_PINB _BV(1)
    #define DAC_PINC _BV(3)
    #define DAC_PIND _BV(4)
    #define DAC_PINS (_BV(2)| _BV(1) | _BV(3) | _BV(4))
    #define LED_PORT PORTL
    #define LED_DDR  DDRL
    #define LED_RX_PIN 3
    #define LED_TX_PIN 5
    #define ADC_PORT PORTF
    #define ADC_DDR  DDRF
    #define ADC_PIN 0
    #define PTT_PORT PORTD
    #define PTT_DDR  DDRD
    #define PPT_PIN 7