markqvist/LibAPRS #17: Telemetry support

danybd, 2019-12-05:
Hello,
First, thank you very much for the library! I want to send the voltage using aprs and I think the proper way is to use the telemetry messages defined in aprs. Are they being supported right now? If not I would try to implement it. Is there any impediment known?

Noah517, 2020-01-14:
For a recent project, I needed to send telemetry as well. I found the easiest way to do this was to convert your value to a string and add it to the comment. If you need help with that I would suggest using the dtostrf() command to convert the value to a string. If you want to include more then one value in the comment or add some extra text you need to make the main comment and substitute the new comment values. Here is a simple example sketch

`
float voltageFloat = voltageValue();

char comment[] = "APRS Test, Voltage: 00";
char voltageString[] = "00";

dtostrf(voltageFloat, 1, 0, voltageString);

comment[20] = voltageString[0];
comment[21] = voltageString[1];

APRS_sendLoc(comment, strlen(comment));
.`

danybd, 2020-08-24:
Thank you for the responses. I'll try the suggestions

closed 2020-08-24 by danybd

4x4eb, 2020-09-25:
hi,
can you share the final telemetry code please?
I have same need and I am looking for some telemetry support.

Eitan